mcci-catena/Modbus-for-Arduino #21: BAD CRC on reading more than 29 registers

dhineshkumarmcci, 2019-01-24:
We were trying to read more registers from a modbus device and found that the library do not support reading data from more than 29 registers.

We gone through the library and seen that **Maximum buffer size(MAX_BUFFER)** is defined as **64** in **ModbusRtu.h**. On reading 29 registers we receive back 63 bytes (Address(1 byte), Function code(1 byte), Data length(1 byte), Register data(58 bytes) and Checksum(2 bytes)). Which is one byte less than the maximum buffer size.

So that we tried reading registers by increasing the **MAX_BUFFER**. But that do not help and not able to read more than 29 registers.

dhineshkumarmcci, 2019-01-28:
Thank you for the input Terry.

I will try increasing the serial ring-buffer size and will change MAX_BUFFER as a parameter in the library. So that user can provide the required buffer size from the sketch.

dhineshkumarmcci, 2019-01-29:
I have tried increasing the ```SERIAL_TX_BUFFER_SIZE``` and ```SERIAL_RX_BUFFER_SIZE``` in "HardwareSerial.h" under STM32 Arduino core. That helped to increase the number of registers to be read. Now we were able to read 72 registers.

On increasing the registers more than 72, we get error code **2**.

terrillmoore, 2019-01-29:
You now know the root cause, please debug. (When posting an error code, please take a moment to look up the symbolic name; 2 is uninformative. Thanks!)

dhineshkumarmcci, 2019-01-29:
The library says that the error code 2 is an exceptional code ```EXC_ADDR_RANGE```. Seems to be an illegal address. Will be debugging further.

Will be making sure to provide all required information in future.

dhineshkumarmcci, 2019-01-30:
We have tried to read 125 registers in the Modbus RTU module. We were able to read all the 125 registers successfully with increasing the hardware serial buffer size.

I will be pushing the changes in BSP for review.

terrillmoore, 2019-01-30:
I believe there's a BSP-independent way to get the serial buffer size -- not checking the `#define` in the STM BSP, but calling `Serial.{something}()`.  Please check, and then modify the library to issue a suitable error message. That way if we run into this again on a different platform, a little research will lead to a comment that references this issue.... Thanks.

commit 48676035 (2019-02-07): Merge pull request #22 from mcci-catena/issue21

Fix #21: Increased maximum buffer size
closed 2019-02-07 by terrillmoore