mcci-catena/Modbus-for-Arduino #35: Cannot compile master: Catena_PollableInterface.h not found

matthijskooijman, 2019-09-30:
I believe a recent (last two weeks) commit broke compilation. After updating the library, I get:

```
In file included from /home/matthijs/docs/Electronics/Arduino/Sketches/libraries/Modbus-for-Arduino/src/lib/cCatenaModbusRtuHost_poll.cpp:17:0:
/home/matthijs/docs/Electronics/Arduino/Sketches/libraries/Modbus-for-Arduino/src/lib/../Catena_ModbusRtuHost.h:21:38: fatal error: Catena_PollableInterface.h: No such file or directory
compilation terminated.
```

This .h file is referenced from inside the library, but not present. Maybe the file was forgotten?

terrillmoore, 2019-09-30:
Compiling that file requires that you also install https://github.com/mcci-catena/Catena-Arduino-Platform.  Of course, not everyone wants to. This problem was introduced when I refactored to pull the `poll()` method implementation out of the `.h` file but this then introduces the dependency. Solution is to move the poll method implementation back to the `.h` file, so if you don't `#include` the Catena wrapper, you won't get this.

Workaround is to remove offending file, as it won't be used unless you also are using the platform.

terrillmoore, 2019-10-01:
@matthijskooijman I compiled several other (non-example) sketches, and I found that I had to be more elaborate in my backward-compatibility approach. New fix pending that seems to work for everything.

terrillmoore, 2019-10-02:
@matthijskooijman if you could temporarily switch to https://github.com/mcci-catena/Modbus-for-Arduino/commit/0460b9fbc081a8cdb2097e77a15604872ee0dd6c (branch issue35b) and test compile, I'd appreciate it.

matthijskooijman, 2019-10-02:
Thanks! Just tested issue35b, which seems to work for my sketch (only compiletested, I do not have hardware at hand to test).

terrillmoore, 2019-10-02:
OK, I'll merge. I've checked all my projects and it seems OK.

commit e8e2788b (2019-09-30): Fix #35: move methods to header file

The Arduino IDE can't optionally compile files based on library
presence, so we can't move methods to files if they need Catena .h
files to compile.

This is a quick hack; the right thing is to either integrate these
methods into the non-Catena class (preferred), or make a separate
library.
commit 0460b9fb (2019-10-02): Fix #35: implement more elaborate/complete backwards compatibility
closed 2019-10-02 by terrillmoore